platformio/platform-atmelsam #156: Update SodaqCore-samd to version 1.8.6

matthias-mueller, 2021-04-08:
It currently sits at `1.6.20` which is severely outdated.

The current stable SodaqCore-samd version is `1.8.6`:

https://forum.sodaq.com/t/sodaqcore-samd-v1-8-6/3342

Although there has been some discussion/work in #93 it seems that no upgrade actually happened.

maxgerhardt, 2021-06-13:
+1 to this -- the branching / tag structure on the https://github.com/SodaqMoja/SodaqCore-samd/ core is also pretty damn confusing in the SODAQ core. The default branch points to 1.6.15. There are no releases on the github, only tags.

closed 2021-06-25 by valeros

matthias-mueller, 2021-06-25:
Thanks for handling this @valeros !

valeros, 2021-06-25:
@matthias-mueller Please retest with the upstream version:
```ini
[env:sodaq_explorer]
platform = https://github.com/platformio/platform-atmelsam.git
framework = arduino
board = sodaq_explorer
```

matthias-mueller, 2021-06-25:
@valeros Sorry to disappint you - but I'm just a "standard" user of PlatformIO and use the version that's visible to the VSCode plugin system.

matthias-mueller, 2021-06-25:
@valeros Compilation works a first tests look okay:

```ini
[env:sodaq_sff]
platform = https://github.com/platformio/platform-atmelsam.git
board = sodaq_sff
framework = arduino
```

keestux, 2021-07-31:
Thanks for the update

commit b2f57a6e (2021-06-25): Update SODAQ SAMD core to the latest v1.8.6

Resolve #156