platformio/platform-atmelsam #164: SAMD51 as Mass storage device using tinyusb

aiweier1231, 2021-07-13:
Hi everyone,
I want to use my Matrix portal m4(SAMD51) as a USB mass storage device with the Circuitpython running in platformio. 
Platformio.init:
[image]

I have selected the USB stack: TinyUsb by adding build_flags = -D USE_TINYUSB.
But somehow, with build_flags = -D USE_TINYUSB. After I upload whatever code,
1. The board no longer shows up under Ports
2. The code that has been uploaded is running fine
3. I can't upload or open the serial monitor.
4. Everything works fine using Arduino IDE to upload.

That would be a great help if someone answers.

Leo

laermansjohan, 2021-07-20:
Hi Leo,

Just wanted to say that I have the same problem. So also very interested in why this problem occurs.

Just for info:

**Board**
Metro M4
**BSP version**
Adafruit SAMD Boards version=1.7.2
**TinyUSB Library version**
1.3.2 - 2021.07.07

**Code**
the basic example demo how to expose on-board external Flash as USB Mass Storage.

**Platformio.ini**
```
[env:adafruit_metro_m4]
platform = atmelsam
board = adafruit_metro_m4
framework = arduino
build_flags = -DUSE_TINYUSB
lib_deps =
    https://github.com/adafruit/SdFat
    https://github.com/adafruit/Adafruit_SPIFlash
```

**What happened ?**
I have a Metro M4 board where the "msc_external_flash" sketch worked (about one year ago) with the Adafruit tinyUSB on platformio and CLion.
After updating pio and the libraries, the board USB functions stopped working.(from tinyUSB 0.10.0, Boards version=1.5.14 -> tinyUSB 1.3.2, Boards version=1.7.2, pio 5.1.1 )

**How to reproduce ?**
I have no compilation errors, however all USB functions of the board do not work, no working msc external flash, nor the serial port that is working. The Tx & Rx LEDs on the Metro stay lid. Other code in the loop runs, the board is not locked.

When I do not set the `build_flags = -DUSE_TINYUSB` in the platformio.ini file and do not use usb_msc (no definition of Adafruit_TinyUSB.h) the boards USB initializes the right way and the serial port works. Also the Tx & Rx LEDs work correct in that case.

**Just to show the versions used:**

PLATFORM: Atmel SAM (6.2.0) Adafruit Metro M4
HARDWARE: SAMD51J19A 120MHz, 192KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, jlink)
PACKAGES:

    framework-arduino-samd-adafruit 1.7.2
    framework-cmsis 2.50400.181126 (5.4.0)
    framework-cmsis-atmel 1.2.2
    toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
    LDF: Library Dependency Finder - http://bit.ly/configure-pio-ldf
    LDF Modes: Finder ~ chain, Compatibility ~ soft
    Found 35 compatible libraries
    Scanning dependencies...
    Dependency Graph
    |-- SdFat - Adafruit Fork 1.2.4+sha.68dd2d4
    | |-- SPI 1.0
    | | |-- Adafruit Zero DMA Library 1.1.0
    | | | |-- Adafruit TinyUSB Library 1.3.2
    | | |-- Adafruit TinyUSB Library 1.3.2
    | |-- Adafruit Zero DMA Library 1.1.0
    | | |-- Adafruit TinyUSB Library 1.3.2
    |-- Adafruit SPIFlash 3.5.2+sha.fa5d4c7
    | |-- SPI 1.0
    | | |-- Adafruit Zero DMA Library 1.1.0
    | | | |-- Adafruit TinyUSB Library 1.3.2
    | | |-- Adafruit TinyUSB Library 1.3.2
    | |-- Adafruit Zero DMA Library 1.1.0
    | | |-- Adafruit TinyUSB Library 1.3.2
    | |-- SdFat - Adafruit Fork 1.2.4+sha.68dd2d4
    | | |-- SPI 1.0
    | | | |-- Adafruit Zero DMA Library 1.1.0
    | | | | |-- Adafruit TinyUSB Library 1.3.2
    | | | |-- Adafruit TinyUSB Library 1.3.2
    | | |-- Adafruit Zero DMA Library 1.1.0
    | | | |-- Adafruit TinyUSB Library 1.3.2
    |-- Adafruit TinyUSB Library 1.3.2
    |-- SPI 1.0
    | |-- Adafruit Zero DMA Library 1.1.0
    | | |-- Adafruit TinyUSB Library 1.3.2
    | |-- Adafruit TinyUSB Library 1.3.2

aiweier1231, 2021-07-20:
Hi there,

I also posted the same problem in the platformio community. Here is the link.
https://community.platformio.org/t/choose-usb-stack-as-tiny-usb/22451/3

In my case, it's the problem with the library linkage issue. I've added **lib_archive = no** in platformio.ini and the problem is resolved.

Hope this would also help you.

laermansjohan, 2021-07-20:
Hi again,

aiweier1231, the `lib_archive = no` in platformio.ini did the trick for me.

Big thanks.
Regards,
Johan