platformio/platform-atmelsam #167: Adafruit feather M4 CAN does not upload

mansk789, 2021-08-05:
there seems to be an issue with the adafruit feather m4 CAN board and uploading. bossac-tool complains "bossac: extra arguments found".

First it seems like the port is not passed correctly to the bossac tool. It passes it as `--port "ttyACM0" `while it should be `--port=ttyACM0`

The second problem I found, there is no offset passed to the bossac-tool. this is due to the main.py L220:
`    if board.get("build.core") in ("adafruit", "seeed", "sparkfun") and board.get(
            "build.mcu").startswith("samd51"):`

In my opinion it should be:
`    if board.get("build.core") in ("adafruit", "seeed", "sparkfun") and (board.get(
            "build.mcu").startswith("samd51") or board.get("build.mcu").startswith("same51")):`

pauljeffress, 2021-08-06:
Looks like may be same as this issue?
https://github.com/platformio/platform-atmelsam/issues/160

mansk789, 2021-08-06:
> Looks like may be same as this issue?
> #160

No this seems to be another issue. They got the error `No device found on COM5
*** [upload] Error 1`

My error is `bossac: extra arguments found`

valeros, 2021-08-09:
Hi @mansk789 , thanks for reporting! I've just pushed a fix and now the offset value should be also used for SAM E51. I believe this should also fix your first problem with the `bossac: extra arguments found` error. Could you please try to run your project using the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:adafruit_feather_m4_can]
platform = https://github.com/platformio/platform-atmelsam.git
board = adafruit_feather_m4_can
framework = arduino
```

mansk789, 2021-08-09:
Hi @valeros 
Works like a charm, thanks for the quick fix! Do you have already a timeline, when the fix will be in the production code? (just for my info, no rush :) )

valeros, 2021-08-10:
Thanks for confirming, a new release will be published by the end of this week.

Resolved in https://github.com/platformio/platform-atmelsam/commit/2c92f8a6a317ac49c7d73c5fcd8d7a5ba1860a28

closed 2021-08-10 by valeros