platformio/platform-atmelsam #175: Feature Request: Add support for Teknic ClearCore Board

bveenema, 2021-09-13:
### Overview
ClearCore is an industrial grade controller board created by Teknic, a motor manufacturer. ClearCore specifically targets industrial controls with 24V tolerant I/O and built-in motor control interfaces.

Teknic has released a CMSIS based motion and I/O library setup for Atmel Studio as well as an Arduino frameworked wrapped that integrates the CMSIS libraries.

ClearCore uses an Atmel SAM ATSAME53N19A processor which is most similar to the existing Adafruit Feather M4 Can processor (SAME51J19A)

### Resources
- [Product Description](https://www.teknic.com/products/io-motion-controller/clcr-4-13/)
- [Manual](https://www.teknic.com/files/downloads/clearcore_user_manual.pdf)
- [Connection Diagram](https://www.teknic.com/files/downloads/clearcore_system_diagram.pdf)
- [ClearCore Library Repository](https://github.com/Teknic-Inc/ClearCore-library)
- [ClearCore Arduino Wrapper Repository](https://github.com/Teknic-Inc/ClearCore-Arduino-wrapper)

### Additional Information
I have not found a concise guide for adding boards to PlatformIO. If one exists, I can do the heavy lifting with support.

thestumbler, 2021-09-20:
I’m currently starting a ClearCore project. This would be an interesting option to have available. The default environment supported by Tecnik is Atmel Studio (now rebadged Microchip Studio) and is seems to be Windows-only.

Lillithiun, 2021-10-01:
I’ve just picked up a project using the ClearCore and direct integration with my preferred IDE would be awesome

Andrea-Loriedo, 2021-10-18:
Same here. Would love PlatformIO support as the default IDE is pretty clunky and Arduino would be pretty limiting for such a powerful controller.

jahasu, 2021-11-07:
Same. Just got mine and would be very keen to get it up and running with platformio

sparky650, 2022-02-02:
I also support the addition of the ClearCore board

jeffrizzo, 2022-03-02:
"me too".  Just got one of these to support a client project, would LOVE to use PlatformIO.

evlquaker, 2022-05-02:
I started making a board file, but cant get it to work. If anyone can help with this it would be much appreciated as I would also like to use PlatformIO with the ClearCore. Below is teknic_clearcore.json

`{
  "build": {
    "arduino":{
      "ldscript": "flash_with_bootloader.ld"
    },
    "cpu": "cortex-m4",
    "extra_flags": [
      "-DARDUINO_CLEARCORE",
      "-D__SAME53N19A__",
      "-D__SAMD51__",
      "-D__FPU_PRESENT",
      "-DARM_MATH_CM4",
      "-DENABLE_CACHE",
      "-DVARIANT_QSPI_BAUD_DEFAULT=50000000"
    ],
    "f_cpu": "120000000L",
    "hwids": [
      [
        "0x2890",
        "0x0022"
      ],
      [
        "0x2890",
        "0x8022"
      ]
    ],
    "mcu": "same53n19a",
    "system": "samd"
  },
  "debug": {
    "jlink_device": "ATSAME53N19A",
    "openocd_chipname": "atsame53N19A",
    "openocd_target": "atsame5x",
    "svd_path": "ATSAME53N19A.svd"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "Teknic ClearCore",
  "upload": {
    "disable_flushing": true,
    "maximum_ram_size": 196608,
    "maximum_size": 507904,
    "native_usb": true,
    "offset_address": "0x4000",
    "protocol": "sam-ba",
    "protocols": [
      "sam-ba",
      "jlink",
      "atmel-ice"
    ],
    "require_upload_port": true,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true
  },
  "url": "https://teknic.com/products/io-motion-controller/",
  "vendor": "Teknic"
}
`

evlquaker, 2022-05-02:
Sorry the formatting was not preserved when I pasted it. :-/

urbhar580, 2022-06-04:
This is off-topic, but it concerns a problem I have with  ClearCore from Teknic. If you have the time, please read my post at:

https://forum.arduino.cc/t/corrupt-board-manager-json-clearcore/999196

buzzysale, 2022-09-29:
I'm willing to contribute USD to help make progress with this.

jacobschleining, 2023-02-08:
Any updates on this? Has anyone made a board file?

thestumbler, 2023-02-08:
I am not familiar with using PlatformIO. However I did recently convert my build process away from Atmel Studio to using G++/ CMAKE. I'm currently setup using GDB with the Atmel ICE, because that's the debugging hardware already installed in my project. But I plan to test with the Black Magic Probe as well. 

If this can help the port-to-PlatformIO project move forward, let me know and I'll post my build files on GitHub.

mrccwolf, 2023-02-23:
+1 this would be super helpful if someone knows how to make the board file

Andrea-Loriedo, 2023-02-23:
> I am not familiar with using PlatformIO. However I did recently convert my build process away from Atmel Studio to using G++/ CMAKE. I'm currently setup using GDB with the Atmel ICE, because that's [...]

I would be very interested in your build files for this, would be super helpful if you put them up!

patrickwasp, 2023-03-09:
I tried flashing a clearcore using the `SparkFun_SAMD51_MicroMod` board definition and it flashed but then the USB port no longer appears. Maybe bricked, or maybe needs to be brought back to life using a debugger.

buzzysale, 2023-03-09:
Like I said I have some $$$ to help pull this off. Trying to port grbl to the Clearcore and having a hell of a time (mechatronics engineer - not enough ee, not enough me)

patrickwasp, 2023-03-09:
> Like I said I have some $$$ to help pull this off. Trying to port grbl to the Clearcore and having a hell of a time (mechatronics engineer - not enough ee, not enough me)

I'll pitch in to get an Upwork freelancer to do the NRE job. I'm not sure what the effort/cost is to add a custom board though.

evlquaker, 2023-03-10:
I looked into this again tonight. It seems that adding the board is more than trivial. I think just making a board.json file is not sufficient as (I don't believe) there is an existing toolchain configured for the Atmel SAM ATSAME53N19A processor. The Adafruit Feather M4 Can board seems to be the only other example of a SAM-E microcontroller. Adafruit uses the SAM-D tool chain. Perhaps its close enough? Configuring the whole toolchain is a bit beyond my current knowledge. I also no longer have a ClearCore board as I installed it in a project for my company. I really hope someone can figure this out.

thestumbler, 2023-03-10:
How tough is mastering Platform IO? As I said, I have the toolchain sorted out for Gcc/G++ but I've never used PlatformIO. I haven't posted a repo yet because I need to test with the bare ClearCore repo, currently it's working for my custom code which I can't release.

evlquaker, 2023-03-10:
I am not sure what you mean by "mastering Platform IO". Do you mean learning enough to make a new board? 

[https://github.com/platformio/platformio-docs/blob/develop/platforms/creating_board.rst](url)

That's a good place to start to learn the basics. 

Then look at Adafruits board as an example

[https://github.com/platformio/platform-atmelsam/blob/develop/boards/adafruit_feather_m4_can.json](url)

The ClearCore board file for Arduino

[https://github.com/Teknic-Inc/ClearCore-Arduino-wrapper/blob/master/package_clearcore_index.json](url)

Then I guess its mainly an issue of figuring out how PlatformIO pulls in the tool chain dependencies. I hope this helps someone

NWClimber, 2023-05-06:
I'm also very interested in adding this board to platform IO. Not sure I can contribute any useful skills here but would greatly appreciate this addition and would gladly test it.

NWClimber, 2023-05-06:
Upvote for adding Teknic ClearCore board.

evlquaker, 2023-05-07:
I did a fair amount of research into what would be required to do this.  I believe it would require adding a Custom Development Platform to PlatformIO in addition to all the board stuff. This is because the existing AtmelSAM Platform [https://github.com/platformio/platform-atmelsam](url)
does not support SAM-E architecture. I don't know what compiler flags are required for this. It could probably be figured out by digging into the Arduino IDE build /upload scripts.

evlquaker, 2023-05-07:
If we can get enough interest in this forum then maybe I can convince Teknic to do it for us ;-)

patrickwasp, 2023-08-28:
With the help of @maxgerhardt I added support for the ClearCore. 

While the pull request is being considered you can use your ClearCore in PlatformIO by using the following config:

```
[env:clearcore]
platform = https://github.com/patrickwasp/platform-atmelsam
board = clearcore
framework = arduino
```