platformio/platform-atmelsam #190: I2C not working on Feather M4 after upload.

jazzlw, 2022-05-07:
I'm new to Platformio, and I've been struggling for the last few hours to get it to properly upload to a Feather M4 and work correctly.

I've finally got it building and uploading but I think something is wrong with the libraries because it will not connect with an I2C device.  In this case I'm trying to use an Adafruit SHT31 board, but I've tried an OLED board and had the same results.  

Currently I'm just trying to build the test sketch that comes with the Adafruit SHT31 library, but when I upload it, the M4 never connects to the sensor and, and the SCL and SDA lines are never pulled down (no I2C communication happens). The Serial.print function does work, and does print to the platformio serial monitor.
 Running the same sketch from the Arduino IDE works just fine.

platformio.ini
```
[env:adafruit_metro_m4]
platform = atmelsam
board = adafruit_metro_m4
framework = arduino
monitor_speed = 115200
lib_deps = 
	adafruit/Adafruit SHT31 Library@^2.1.0
	adafruit/Adafruit BusIO@^1.11.5
	adafruit/Adafruit Zero DMA Library @ ^1.0.8
	SPI
	Wire
```

[220507-162823-adafruit_metro_m4.zip](https://github.com/platformio/platform-atmelsam/files/8646074/220507-162823-adafruit_metro_m4.zip)

pauljeffress, 2022-05-08:
Could it be that you have selected the Adafruit Metro M4 and not the
Adafruit Feather M4 as your board type?

I use i2c on Feather m4 fine.

Am not near a machine to give you all the details but I know it works.

Cheers,
PJ

pauljeffress, 2022-05-08:
I actually use a Feather M4 CAN board and here is my platform ini

[env:adafruit_feather_m4_can]
platform = atmelsam
board = adafruit_feather_m4_can
framework = arduino
upload_port = /dev/cu.usbmodem11301
monitor_port = /dev/cu.usbmodem11301
monitor_speed = 115200
monitor_filters = time, log2file, colorize
lib_deps = 
	adafruit/Adafruit HX8357 Library@^1.1.10
	adafruit/Adafruit BusIO@^1.7.2
	adafruit/Adafruit GFX Library@^1.10.7
	paulstoffregen/OneWire@^2.3.5
	milesburton/DallasTemperature@^3.9.1
	paulstoffregen/Time@^1.6
	powerbroker2/SerialTransfer@^3.1.1
	khoih-prog/SAMD_TimerInterrupt@^1.4.0
	adafruit/CAN Adafruit Fork@^1.2.1
	sparkfun/SparkFun u-blox GNSS Arduino Library@^2.2.7
	adafruit/RTClib@^2.0.2
lib_extra_dirs = /Users/paul/Dropbox/Electronics/Arduino/PJ_PlatformIO_Projects/Pulsar_Shared_Source/

jazzlw, 2022-05-08:
wow, thanks @pauljeffress , I can't believe  I didn't notice that I'd selected the wrong board... 🤦 🤦    I'm not near the board at the moment but that seems like a very good explanation.

Thanks again.

closed 2022-05-08 by jazzlw