platformio/platform-atmelsam #192: Debug and serial monitor

fterretf, 2022-05-12:
Using the serial monitor works fine when doing upload. I just do platformIO:upload and the serial monitor opens and messages are displayed.

With the mkr1010wifi powered by USB cable connected to a 5 V power supply, debugger is fine. I can step through my program but no serial port.

With the mkr1010wifi powered to a PC USB port, the debugger works fine until USBDevice.attach() is executed.
[image]

At this point, windows10 send a ‘USB device not connected’ and the windows device manager gives error code 43

[image]

As suggested by Max Gerhardt, I add in  platformio.ini the initial breakpoint after the USBdevice.attach() and compilation with optimization, but this doesn't help.
debug_build_flags = -Os -g -ggdb
debug_init_break = break setup

ivankravets, 2022-05-19:
Could you:

1. Remove `debug_build_flags = -Os -g -ggdb` line
2. Replace `debug_init_break = break setup`  with `debug_init_break = tb setup`

Does it stop at setup now?

fterretf, 2022-05-20:
See workaround
[Cannot debug ledControl example of ArduinoBLE on mkr1010 with jlink plus probe](https://community.platformio.org/t/cannot-debug-ledcontrol-example-of-arduinoble-on-mkr1010-with-jlink-plus-probe/27768)