platformio/platform-atmelsam #193: Seeeduino XIAO - upload fails at about 50% of the time

mightycoco, 2022-05-16:
Uploading a cpp project to Seeeduino XIAO failes at about 50% of the time. The bossac executable stops after "Uploading...":

```
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM11
Forcing reset using 1200bps open/close on port COM11
Waiting for the new upload port...
Uploading .pio\build\seeed_xiao\firmware.bin
```
a working upload looks like this 
```
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM11
Forcing reset using 1200bps open/close on port COM11
Waiting for the new upload port...
Uploading .pio\build\seeed_xiao\firmware.bin
Atmel SMART device 0x10010005 found
Erase flash
done in 0.819 seconds

Write 76248 bytes to flash (1192 pages)

[=                             ] 5% (64/1192 pages)
[===                           ] 10% (128/1192 pages)
[====                          ] 16% (192/1192 pages)
[======                        ] 21% (256/1192 pages)
[========                      ] 26% (320/1192 pages)
[=========                     ] 32% (384/1192 pages)
[===========                   ] 37% (448/1192 pages)
[============                  ] 42% (512/1192 pages)
[==============                ] 48% (576/1192 pages)
[================              ] 53% (640/1192 pages)
[=================             ] 59% (704/1192 pages)
[===================           ] 64% (768/1192 pages)
[====================          ] 69% (832/1192 pages)
[======================        ] 75% (896/1192 pages)
[========================      ] 80% (960/1192 pages)
[=========================     ] 85% (1024/1192 pages)
[===========================   ] 91% (1088/1192 pages)
[============================  ] 96% (1152/1192 pages)
[==============================] 100% (1192/1192 pages)
done in 0.570 seconds

Verify 76248 bytes of flash with checksum.
Verify successful
done in 0.094 seconds
CPU reset.
```

The upload always works if 

- The device was put into programming mode manually (cumbersom as a pin has to be pulled to ground in a sequence)
- The device was put into programming mode when manually calling "mode com10: BAUD=1200" from the command line

Prior uploading the device is put to programming mode in pioupload.py using TouchSerialPort which works only about 50% of the time.

I never had issues with uploading on my old PC but after having to upgrade to a new workstation, the situation is as described.

m-r-m-s, 2022-08-17:
@mightycoco thank you for reporting this issue! I thought this was a "quirk" of the XIAO and not related to Platformio. Having to manually put XIAO into bootloader mode with the pins is as you say very cumbersome! I experience a high upload failure rate quite often as well when using Platformio.

Do you get different / better success rates from Arduino IDE?

mightycoco, 2022-08-17:
> @mightycoco thank you for reporting this issue! I thought this was a "quirk" of the XIAO and not related to Platformio. Having to manually put XIAO into bootloader mode with the pins is as you say is very cumbersome! I experience a high upload failure rate quite often as well when using Platformio.
> 
> Do you get different / better success rates from Arduino IDE? 

Haven't tried using arduino ide lately. But a good point, I need to test out.

Currently I unplug the device and replug after the compiler is done. That way, the upload works more often than not.

The thing is, I've rarely experienced this on my old amd machine. So I guess it has something to do with my hardware configuration.

besi, 2023-06-15:
I'm experiencing the same problem using the latest PlatformIO. When the upload fails I get the error message:

    Forcing reset with 1200bps

monroewilliams, 2024-12-31:
I'm also having this problem with Seeeduino XIAO and Adafruit QT Py M0 boards (very similar designs, both SAMD21). I'm running PlatformIO under VS Code on Mac OS X.

I was able to fix it locally by building my own bossac from the 1.7.0 source tag with the patch from this PR applied: https://github.com/shumatech/BOSSA/pull/150

To build the 1.7.0 bossac sources, I did have to fix up the OSX section of the makefile (removing -arch i386 and -mmacosx-version-min=10.5 (!) from COMMON_CXXFLAGS/COMMON_LDFLAGS and adding -arch arm64).

siganberg, 2025-01-01:
I read that this happen on MacOS because during brief time of upload, the MacOS is trying to map it as external storage device which messses up the firmware upload.  The solution was to add UUID of the Seeed to the exception of auto mount. This work on my old MacOS with intel but the same solution somehow still doesn't work with latest MacOS or with silicone chip.