platformio/platform-atmelsam #198: Adafruit's TinyUSB does not compile on XIAO (Anniversary Edition)

DP1000, 2022-10-05:
Hi everyone, i'ts almost a year that Adafruit's TinyUSB does not compile on Seeduino XIAO when using PlatformIO.
Tried again now with `Atmel SAM@8.1.0` and `Adafruit TinyUSB@1.14.4`.
Adafruit says that the problem is yours because on Arduino IDE the library works great.
I'm not the only one that has this problem, Adafruit TinyUSB repo's full of issues like this, all closed without a fix.
Please don't ignore this issue [like the other one that i made almost a year ago](https://github.com/platformio/platform-atmelsam/issues/179), thanks.

freasy, 2022-10-31:
Bump - having the same problem!

pwfff, 2023-01-21:
I think the issue is actually in this repo: https://github.com/Seeed-Studio/ArduinoCore-samd/issues

The TinyUSB submodule there is ancient. I tried updating it myself but I can't even manage to get a custom platform installed in platformio :(

If anyone knows how I could actually test my fork, let me know.

DP1000, 2023-01-23:
Hi @pwfff should we open an issue at ArduinoCore-samd ?

Probably you already had, but have you looked at [platformio docs](https://docs.platformio.org/en/stable/platforms/creating_platform.html) ?

Theremingenieur, 2023-11-06:
It seems like Seeedstudio's ArduinoCore-samd has been updated to v. 1.8.5 on September 4, 2023 and this might fix some longstanding issues with Adafruit's TinyUSB. They even pack/link a specific version into their package. https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/6d82f2263b6dfb822433e9cd2328657715c6b3f4
@ivankravets : would it be possible to update samd-seeed accordingly to v.1.8.5 please ? I'd then check out if that solves the issue. Thank you in advance !