platformio/platform-atmelsam #209: Programming ATSAME51J19A

lillikelley, 2023-06-15:
Hello all,

I am trying to use the Black Magic Probe to load firmware (.hex w/ bootloader) onto an ATSAME51J19A using a SWD header on a separate PCB. I have updated my BMP firmware as well as the PIO core (I am using the VSCode extension but am totally fine using the CLI). I have my board set to the Adafruit feather w/ can since they have the same MCU, even though they're not the same board, but let me know if that's the root of my problems here. I have edited my C:\Users\<user>\.platformio\platforms\atmelsam\boards\adafruit_feather_m4_can.json file and added "blackmagic" into the protocols array to no avail. 

Here is my platformio.ini file:

[env:adafruit_feather_m4_can]
platform = [GitHub - platformio/platform-atmelsam: Atmel SAM: development platform for PlatformIO 1](https://github.com/platformio/platform-atmelsam)
board = adafruit_feather_m4_can
framework = arduino
debug_tool = blackmagic
debug_port = COM4
upload_port = COM4
upload_protocol = blackmagic

valeros, 2023-06-16:
Hi @lillikelley, what's the error message exactly? Anyway, the `adafruit_feather_m4_can` board seems to have a bootloader, so it cannot be used with a generic `ATSAME51J19A ` board.

lillikelley, 2023-06-16:
Hi, thank you for answering! Would it be easy to make a custom .json for the SAME51 chip using some of the information from the feather w/ can board? Here is my original error message:

Verbose mode can be enabled via `-v, --verbose` option
DebugInvalidOptionsError: Unknown debug tool `blackmagic`. Please use one of `atmel-ice, jlink` or `custom`:
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\lkelley\.platformio\platforms\atmelsam@src-3902fa010271b7fd4b3fe129967364bc\builder\main.py", line 127:
    target_elf = env.BuildProgram()
  File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 110:
    env.PrintConfiguration()
  File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\tools\pioplatform.py", line 233:
    _get_debug_data(),
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\tools\pioplatform.py", line 197:
    % board_config.get_debug_tool_name(env.GetProjectOption("debug_tool")),
  File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\platform\board.py", line 127:
    raise DebugInvalidOptionsError(
========================== [FAILED] Took 1.01 seconds ==========================

lillikelley, 2023-06-16:
Alright, so this is as far as I got after making those changes. The project built successfully, and I have the .hex file to be programmed in `.pio/build/adafruit_feather_m4_can/`. Is my next step to make a custom linker script for the SAME51 chip specifically? This is my new error message:

 *  Executing task: C:\Users\lkelley\.platformio\penv\Scripts\platformio.exe run --target upload --environment adafruit_feather_m4_can 

Processing adafruit_feather_m4_can (platform: https://github.com/platformio/platform-atmelsam; board: adafruit_feather_m4_can; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/adafruit_feather_m4_can.html
PLATFORM: Atmel SAM (8.1.0+sha.61f7971) > Adafruit Feather M4 CAN
HARDWARE: SAME51J19A 120MHz, 192KB RAM, 496KB Flash
DEBUG: Current (blackmagic) External (atmel-ice, blackmagic, jlink)
PACKAGES:
 - framework-arduino-samd-adafruit @ 1.7.10
 - framework-cmsis @ 2.50400.181126 (5.4.0)
 - framework-cmsis-atmel @ 1.2.2
 - tool-openocd @ 3.1200.0 (12.0)
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\adafruit_feather_m4_can\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.4% (used 2664 bytes from 196608 bytes)
Flash: [          ]   2.2% (used 11252 bytes from 507904 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = blackmagic
Looking for BlackMagic port...
Using manually specified: COM4
Uploading .pio\build\adafruit_feather_m4_can\firmware.bin
C:\Users\lkelley\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Microchip SAME51N20A (rev D) M4
0xfffffffe in ?? ()
Error erasing flash with vFlashErase packet
Section .text, range 0x0 -- 0x2af4: MIS-MATCHED!
warning: One or more sections of the target image does not match
Section .data, range 0x2af4 -- 0x2bf4: MIS-MATCHED!
the loaded file
Kill the program being debugged? (y or n) [answered Y; input not from terminal]

[Inferior 1 (Remote target) killed]
================================================================================================ [SUCCESS] Took 3.39 seconds ================================================================================================

pauljeffress, 2023-06-16:
Hi @lillikelley 
While I don’t know much about this sort of thing, I just happened to notice this line in your output

1 Microchip SAME51N20A (rev D) M4

Is that an indicator of the issue?  It’s not the ATSAME51J19A part?

lillikelley, 2023-06-20:
Hi @valeros, I ran the "Project Inspect" feature and found that there is an invalid board option 'upload.offset_address'. My firmware includes the bootloader, but the `flash_without_bootloader.ld` file has the build offset, so there's a mismatch. Here is my error output after inspecting memory and checking code on my debug environment:

PIO Core Call Error: "Processing feather_can_debug (platform: https://github.com/platformio/platform-atmelsam; board: adafruit_feather_m4_can; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/adafruit_feather_m4_can.html
PLATFORM: Atmel SAM (8.1.0+sha.61f7971) > Adafruit Feather M4 CAN
HARDWARE: SAME51J19A 120MHz, 192KB RAM, 496KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd-adafruit @ 1.7.10 
 - framework-cmsis @ 2.50400.181126 (5.4.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode

KeyError: "Invalid board option 'upload.offset_address'":
 File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:
 env.SConscript("$BUILD_SCRIPT")
 File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
 return _SConscript(self.fs, *files, **subst_kw)
 File "C:\Users\lkelley\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
 exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
 File "C:\Users\lkelley\.platformio\platforms\atmelsam\builder\main.py", line 224:
 "-U", "--offset", board.get("upload.offset_address")])
 File "C:\Users\lkelley\.platformio\penv\Lib\site-packages\platformio\platform\board.py", line 47:
 raise KeyError("Invalid board option '%s'" % path)
========================== [FAILED] Took 1.22 seconds ==========================
Environment Status Duration
----------------- -------- ------------
feather_can_debug FAILED 00:00:01.218
==================== 1 failed, 0 succeeded in 00:00:01.218 ====================

valeros, 2023-06-20:
> Here is my error output after inspecting memory and checking code on my debug environment:

You're probably using an incorrect option name, it should be `board_upload.offset_address` not `upload.offset_address`.