platformio/platform-atmelsam #210: EEPROM useage like ESPs do

nerdyscout, 2023-06-16:
this is more a question than an issue:
most ESP come with 4MB flash, while uploading files from the folder `data` while be stored there and after on accessible via SPIFFS or LittleFS. Boards like the @Adafruit [Metro M0 Express](https://learn.adafruit.com/adafruit-metro-m0-express) got a SPI Flash as well. 
**What is needed to use this in a similar manner?**
I need to store some project configuration (*.json) file and that would be the ideal way... could you please point my into the correct direction? I am happy to implement needed stuff by myself... I am just wondering why this has not been done with similar boards.

valeros, 2023-06-19:
Hi @nerdyscout  The atmelsam platform is lacking necessary tools to reproduce similar to ESP32 functionality. AFAIK, the only way is to use the [Adafruit SPIFlash](https://registry.platformio.org/libraries/adafruit/Adafruit%20SPIFlash) library. Here are [some examples](https://github.com/adafruit/Adafruit_SPIFlash/tree/master/examples) and a [tutorial](https://learn.adafruit.com/adafruit-metro-m0-express/using-spi-flash).

closed 2023-06-19 by valeros

nerdyscout, 2023-06-19:
> Hi @nerdyscout The atmelsam platform is lacking necessary tools to reproduce similar to ESP32 functionality. AFAIK, the only way is to use the [Adafruit SPIFlash](https://registry.platformio.org/libraries/adafruit/Adafruit%20SPIFlash) library. 

Yes, I am aware of this. I was more asking WHY they are lacking... is there anything from blocking an implementation? I think it is the much better way to go, but so far I was not able to figure out how those things work together.

valeros, 2023-06-19:
> Yes, I am aware of this. I was more asking WHY they are lacking... is there anything from blocking an implementation? I think it is the much better way to go, but so far I was not able to figure out how those things work together.

This dev-platform only implements already available functionality. The Espressif community provides all necessary tools, software and workflows while Arduino/Microchip/Adafruit/etc. team decided to confine themselves to a simple library. So this repository is not quite the right place to ask `WHY` questions. You'd better communicate with https://github.com/adafruit/ArduinoCore-samd or any other SAMD community.