platformio/platform-atmelsam #227: Release 8.3.0 breaks my built on Metro-M4

RudolphRiedel, 2024-06-23:
I just tried to build my Arduino example project:
https://github.com/RudolphRiedel/FT800-FT813/tree/5.x/examples/EVE_Test_Arduino_PlatformIO

And it builds for everything that is configured except adafruit_metro_m4.
Using platform = atmelsam@8.2.1 it builds again, so something in 8.3.0 broke it.

It might be the update of framework-arduino-samd-adafruit from 1.7.13 to "1.10716.0".
But it works fine for the two SAMD21 boards in the example that also list framework-arduino-samd-adafruit as dependancy.

It starts failing over Adafruit_TinyUSB_API.cpp
*** [.pio\build\adafruit_metro_m4\lib404\Adafruit_TinyUSB_Arduino\arduino\Adafruit_TinyUSB_API.cpp.o] Error 1
In file included from C:\Users\Ich\.platformio\packages\framework-arduino-samd-adafruit\cores\arduino/Arduino.h:157,
                 from C:\Users\Ich\.platformio\packages\framework-arduino-samd-adafruit\libraries\Adafruit_TinyUSB_Arduino\src\arduino\Adafruit_USBD_CDC.cpp:30:
C:\Users\Ich\.platformio\packages\framework-arduino-samd-adafruit\cores\arduino/USB/USBCore.h:119:32: error: expected identifier before numeric constant
  119 | #define MSC_SUBCLASS_SCSI      0x06
      |                                ^~~~
c:\users\ich\.platformio\packages\framework-arduino-samd-adafruit\libraries\adafruit_tinyusb_arduino\src\class/msc/msc.h:47:3: note: in expansion of macro 'MSC_SUBCLASS_SCSI'
   47 |   MSC_SUBCLASS_SCSI      ///< SCSI transparent command set
      |   ^~~~~~~~~~~~~~~~~
C:\Users\Ich\.platformio\packages\framework-arduino-samd-adafruit\cores\arduino/USB/USBCore.h:119:32: error: expected '}' before numeric constant
  119 | #define MSC_SUBCLASS_SCSI      0x06

And I am not even using USB, or at least I am not trying to.

I found this: https://github.com/earlephilhower/arduino-pico/issues/477
But it was resolved two years ago.
And yes, I am using SPI.

Adding the suggested lib_ldf_mode = chain+ to my platformio,ini worked for me and the example builds for all boards now, but the question really is - what broke this time and why?

valeros, 2024-06-24:
Hi @RudolphRiedel. I'd recommend sticking to the `chain+` mode which emulates Arduino IDE behavior. The reason why it happens is because the `Adafruit Zero DMA` has an optional include from the `Adafruit TinyUSB` library and hence makes PlatformIO think it's an implicit dependency.

RudolphRiedel, 2024-06-24:
Yes, I can build.
The question still is, what broke it after it was already fixed before?

rcross-lc, 2024-07-11:
Also experiencing this, built fine before.

Divide-And-Conquer, 2024-08-19:
I'm also suffering from this problem, the lib_ldf_mode = chain+ works, but I'd still like to see it fixed.