platformio/platform-atmelsam #234: Release 8.1.0 (and newer) broke on custom board without USB

MSmeets94, 2024-10-28:
We have a custom board without USB using release 8.1.0. Because we don't have USB, we build with 'build_unflags = -DUSBCON'.

platform-atmelsam targets framework-arduino-samd ~1.8.13 which was never a problem until 1.8.14 came out. 

Since the release of framework-arduino-samd 1.8.14, a LineInfo declaration was added to the Serial_ class in USBAPI.h which causes a compilation error because the definition of LineInfo is in CDC.h (which is not defined because of the unbuild flag).

valeros, 2024-10-29:
Hi @MSmeets94! It seems you deliberately removed a flag that's required in the latest version of the Arduino version, thus I'm not sure PlatformIO has something to do with the issue. As a workaround, you can explicitly lock the Arduino package to the last known working version using the [platform_packages](https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform_packages.html) option, for example:

```ini
[env:zero]
platform = atmelsam
framework = arduino
board = zero
platform_packages =
  platformio/framework-arduino-samd @ 1.8.13
```

MSmeets94, 2024-10-30:
Hi @valeros! 

Thank you for your reaction. As far as I know, ‘USBCON’ is or was only necessary when using the Arduino USB API and it was correct to undefine it when no USB API is needed. I agree that this might cause issues with newer Arduino versions which is understandable. However, I believe the problem isn’t with the Arduino Core or version itself, but rather with my build suddenly failing despite no changes to dependencies or versions.

I do expect that while using the same dependency versions, i am able to deterministically build the project. I think the underlying issue here is actually in 'framework-arduino-samd', because there is a breaking change in a patch version (between 1.8.13 and 1.8.14). 

According to the release notes for platform-atmelsam 8.3.0, the Arduino Core was updated to 1.8.14, which is fine of course. However, platform-atmelsam 8.1.0 also implicitly updates to the same Arduino Core version, even though its release notes state that the Arduino Core version for this release is 1.8.13. The same is actually true for 'framework-arduino-samd-seeed' which was excplitly bumped in 8.3.0, but if used in 8.1.0 would also get bumped to the same version. 

I do understand this is kind of a niche issue and the work around does work, but still the dependency versions of this project should not change in my opinion, if a release is created to explicitly bump these versions. Looking at release 8.1.0, the `~` in the dependencies in package.json allows for this implicit update between these patch versions. 

 ```
"framework-arduino-samd": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": "~1.8.13"
    },
```

valeros, 2024-10-30:
I completely understand your frustration, but unfortunately, we have no control over Arduino releases, so we can only rely on the sanity of the Arduino core maintainers. If they release a new patch version we expect that there are no breaking changes, that's why the version range for the `framework-arduino-samd` package is configured to accept implicit patch updates.

MSmeets94, 2024-10-30:
I completely understand, i was just wondering because the release notes explictly bump the (patch) versions. I opened an issue in Arduino Core for this.