platformio/platform-atmelsam #237: Variants support for custom boards

abetis, 2025-02-13:
Hi all.

I think I've read all the tutorials and discussions about adding a custom board to platformio.
Specifically this discussion was VERY helpful:
https://community.platformio.org/t/custom-nrf52832-board-pinout/24491

There is a "project" level support, where I can add "boards" and "variants" directories,
add custom_board.json to the boards direcory and variant specific files to the variants directory.
In the custom_board.json I can add the "variants_dir" parameter that points to relative "variants" directory.
That option works great.

But, there is only a partial support for global level configuration...
I can have the "boards" directory in my ".platformio" (I'm with Linux) with the custom_board.json file, which will be recognized by the platform and can be selected from the UI for new projects.
BUT... Looks like the scripts won't look for the relative "variants" directory in the ".platformio", as specified by the "variants_dir" parameter.

Can that part of the needed support be added somewhere in the scipts please?
I'm not familiar with the platformio code, but I believe it shouldn't be a big problem.
Probably copy the part of the code that handles the global boards directory recognition and add similar processing for the directory specified in the variants_dir option.

To make it quick and dirty, we could say that the variants_dir option should always point to the relative "variants" directory, similar to the hardcoded "boards" directory in the ".platformio".

If someone can point me to the correct place in the code, I might try to add that support myself and push a PR.

Without that support, users need to add "build_flags = -I /full-path-to-variants-directory" to the platform.ini file, which is not that convenient if you try to add a global support for all the projects...
Even if you add it to the ini, there are still linking problems that look for the "flash_with_bootloader.ld" file, which have to be resolved...

Adding the same support for the global variants directory, similar to the global boards directory will solve all those problems...

Thanks.

valeros, 2025-02-17:
Hi @4project-co-il, a PR is always welcome, the code responsible for selecting a proper variant directory for SAMD targets is located [here](https://github.com/platformio/platform-atmelsam/blob/develop/builder/frameworks/arduino/arduino-samd.py#L181-L183) and [here](https://github.com/platformio/platform-atmelsam/blob/develop/builder/frameworks/arduino/arduino-common.py#L98-L100).

4project-co-il, 2025-02-17:
on it...

4project-co-il, 2025-02-18:
#238

4project-co-il, 2025-02-18:
@valeros 
Will those changes work only from the platformio.ini?
Or I can set the `variants_dir` in the custom board `.json` file as well?
What should I set it to to make sure the `variants` under the `core_dir` directory is used?
Should it be `"variants_dir" : "${platformio.core_dir}/variants"`?
It doesn't work for me :(

4project-co-il, 2025-02-18:
@valeros 
Oh, I see... If I set it to `"variants_dir" : "$PROJECT_CORE_DIR/variants"` it works!

The **PROJECT** in the `PROJECT_CORE_DIR` is a bit missleading. Thought it relates somehow to the project directory.

Thanks for the change!

Also, please see that issue from 2020, maybe you can push it somehow:
https://github.com/platformio/platform-ststm32/issues/462

valeros, 2025-02-19:
> Or I can set the variants_dir in the custom board .json file as well?

`platformio.ini` options that start with `board_` basically override values in the `.json` board manifest.  
`${platformio.core_dir}` will work only in the `platformio.ini`, while `$PROJECT_CORE_DIR` should work in both cases.

valeros, 2025-02-19:
Resolved in #238