platformio/platform-atmelsam #240: Compiler Error in framework-arduino-samd-clearcore

vkapadia70, 2025-04-25:
Getting compiler error while compiling with:
platform = atmelsam
board = clearcore
framework = arduino

I get the following error while buidling ArduinoJson library:

In file included from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/ArduinoString.h:32:0,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/Print.h:24,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/Stream.h:24,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/Client.h:22,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/ArduinoAPI.h:29,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino/Arduino.h:24,
                 from C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/SPI.h:23,
                 from src\SDcard.cpp:4:
.pio\libdeps\clearcore_hvac\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'const T* ArduinoJson::V741PB22::detail::pgm_read(const T* const*) [with T = char]':
.pio\libdeps\clearcore_hvac\ArduinoJson\src/ArduinoJson/Deserialization/DeserializationError.hpp:85:42:   required from here
C:\Users\admin\.platformio\packages\framework-arduino-samd-clearcore\cores\arduino\api/deprecated-avr-comp/avr/pgmspace.h:106:29: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))

I had to modify the line to:
 #define pgm_read_ptr(addr) ((const void *)(addr)) to fix this error.

valeros, 2025-04-25:
Hi @vkapadia70, the `atmelsam` dev-platform doesn't support this core.

closed 2025-04-25 by valeros

vkapadia70, 2025-04-25:
Sorry, I needed to raise this on https://github.com/patrickwasp/platform-atmelsam.